STMicroelectronics/STMems_Linux_IIO_drivers #2: st_lsm6dsx Industrial IO IMU driver not loading

Panky-codes, 2020-04-13:
I have  LSM6DS3 IMU sensor from ST. I was successful in writing a userspace SPI driver to interact with this sensor from my beaglebone black, and now I am trying to use an IIO kernel module driver, st_lsm6dsx_spi that comes pre-built. 
However, even after countless hours of debugging, I am having trouble getting it to work from kernel space.

I adapted my overlay (posted below) to use this LKM for spi0. I can see a device being created in /sys/bus/spi as follows:
```
debian@beaglebone:~$ cat /sys/bus/spi/devices/spi1.0/
modalias    of_node/    power/      statistics/ subsystem/  uevent
debian@beaglebone:~$ cat /sys/bus/spi/devices/spi1.0/modalias 
spi: lsm6ds3
```
But I don't see the correct driver is loaded for it. It not creating any char device in `/dev/` directory nor I can see any sign of it in `/sys/bus/iio/devices` directory.
No effect even if I try to manually load the driver, st_lsm6dsx_spi using modprobe. 

Could someone say what is going on and what I am doing wrong here? Any help would be highly appreciated. 

**More information:**
For now, I don't want to enable interrupts for fifo mode. 
My DTS overlay: 
```
  /dts-v1/;
/plugin/;

#include <dt-bindings/board/am335x-bbw-bbb-base.h>
#include <dt-bindings/pinctrl/am33xx.h>

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black";

	/* identification */
    part-number = "BB-LSM6DX";
	version = "00A0";

	/* state the resources this cape uses */
	exclusive-use =
		/* the pin header uses */
		"P9.17",	/* P9_17 (A16) spi0_cs0.spi0_cs0 */
		"P9.18",	/* P9_18 (B16) spi0_d1.spi0_d1 */
		"P9.21",	/* P9_21 (B17) spi0_d0.spi0_d0 */
		"P9.22",	/* P9_22 (A17) spi0_sclk.spi0_sclk */
		/* the hardware ip uses */
		"spi0";

	/*
	 * Helper to show loaded overlays under: /proc/device-tree/chosen/overlays/
	 */
	fragment@0 {
		target-path="/";
		__overlay__ {

			chosen {
				overlays {
					BB-LSM6DSX-00A0 = __TIMESTAMP__;
				};
			};
		};
	};

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@1 {
		target = <&ocp>;
		__overlay__ {
			P9_17_pinmux { status = "disabled"; };	/* P9_17 (A16) spi0_cs0.spi0_cs0 */
			P9_18_pinmux { status = "disabled"; };	/* P9_18 (B16) spi0_d1.spi0_d1 */
			P9_21_pinmux { status = "disabled"; };	/* P9_21 (B17) spi0_d0.spi0_d0 */
			P9_22_pinmux { status = "disabled"; };	/* P9_22 (A17) spi0_sclk.spi0_sclk */
		};
	};

	fragment@2 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_spi0_pins: pinmux_bb_spi0_pins {
				pinctrl-single,pins = <
					AM33XX_PADCONF(AM335X_PIN_SPI0_SCLK, PIN_INPUT, MUX_MODE0) /* P9_22 (A17) spi0_sclk.spi0_sclk */
					AM33XX_PADCONF(AM335X_PIN_SPI0_D0, PIN_INPUT, MUX_MODE0)   /* P9_21 (B17) spi0_d0.spi0_d0 */
					AM33XX_PADCONF(AM335X_PIN_SPI0_D1, PIN_INPUT, MUX_MODE0)   /* P9_18 (B16) spi0_d1.spi0_d1 */
					AM33XX_PADCONF(AM335X_PIN_SPI0_CS0, PIN_INPUT, MUX_MODE0)  /* P9_17 (A16) spi0_cs0.spi0_cs0 */
				>;
			};
		};
	};
	fragment@3 {
		target = <&spi0>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;

			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_spi0_pins>;
            channel@0 {
                    status = "disabled"; /* Enabled by default in bb-kernel so users can use spidev from userspace */
            };
            channel@1 {
                    status = "disabled";
            };
           lsm6ds3: lsm6ds3@0 {
				spi-max-frequency = <500000>;
				compatible = "st, lsm6ds3";
                                reg = <0x0>;
			};

		};
	};
};

```

My `sudo /opt/scripts/tools/version.sh` output:
```
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLT00C05014BBBK1F75]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-LSM6DSX-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.114108] remoteproc remoteproc0: wkup_m3 is available
[    1.322488] remoteproc remoteproc0: powering up wkup_m3
[    1.322607] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.326544] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.872348] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.873576] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

```

mariotesi, 2020-04-14:
Can we see the kernel log after boot or after modprobe ?
You can check if there are some kernel messages about lsm6ds3 device ?

Panky-codes, 2020-04-14:
Hi @mariotesi. Thanks for your reply. I just found that it was an extra space (" ")for the compatible key that was creating the problem.
 
My final DT overlay that works is as follows:
``` /dts-v1/;
/plugin/;

#include <dt-bindings/board/am335x-bbw-bbb-base.h>
#include <dt-bindings/pinctrl/am33xx.h>

/ {
	compatible = "ti,beaglebone", "ti,beaglebone-black";

	/* identification */
    part-number = "BB-LSM6DX";
	version = "00A0";

	/* state the resources this cape uses */
	exclusive-use =
		/* the pin header uses */
		"P9.17",	/* P9_17 (A16) spi0_cs0.spi0_cs0 */
		"P9.18",	/* P9_18 (B16) spi0_d1.spi0_d1 */
		"P9.21",	/* P9_21 (B17) spi0_d0.spi0_d0 */
		"P9.22",	/* P9_22 (A17) spi0_sclk.spi0_sclk */
		/* the hardware ip uses */
		"spi0";

	/*
	 * Helper to show loaded overlays under: /proc/device-tree/chosen/overlays/
	 */
	fragment@0 {
		target-path="/";
		__overlay__ {

			chosen {
				overlays {
					BB-LSM6DSX-00A0 = __TIMESTAMP__;
				};
			};
		};
	};

	/*
	 * Free up the pins used by the cape from the pinmux helpers.
	 */
	fragment@1 {
		target = <&ocp>;
		__overlay__ {
			P9_17_pinmux { status = "disabled"; };	/* P9_17 (A16) spi0_cs0.spi0_cs0 */
			P9_18_pinmux { status = "disabled"; };	/* P9_18 (B16) spi0_d1.spi0_d1 */
			P9_21_pinmux { status = "disabled"; };	/* P9_21 (B17) spi0_d0.spi0_d0 */
			P9_22_pinmux { status = "disabled"; };	/* P9_22 (A17) spi0_sclk.spi0_sclk */
		};
	};

	fragment@2 {
		target = <&am33xx_pinmux>;
		__overlay__ {
			bb_spi0_pins: pinmux_bb_spi0_pins {
				pinctrl-single,pins = <
					AM33XX_PADCONF(AM335X_PIN_SPI0_SCLK, PIN_INPUT, MUX_MODE0) /* P9_22 (A17) spi0_sclk.spi0_sclk */
					AM33XX_PADCONF(AM335X_PIN_SPI0_D0, PIN_INPUT, MUX_MODE0)   /* P9_21 (B17) spi0_d0.spi0_d0 */
					AM33XX_PADCONF(AM335X_PIN_SPI0_D1, PIN_INPUT, MUX_MODE0)   /* P9_18 (B16) spi0_d1.spi0_d1 */
					AM33XX_PADCONF(AM335X_PIN_SPI0_CS0, PIN_INPUT, MUX_MODE0)  /* P9_17 (A16) spi0_cs0.spi0_cs0 */
				>;
			};
		};
	};
	fragment@3 {
		target = <&spi0>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;

			status = "okay";
			pinctrl-names = "default";
			pinctrl-0 = <&bb_spi0_pins>;
            channel@0 {
                    status = "disabled"; /* Enabled by default in bb-kernel so users can use spidev from userspace */
            };
            channel@1 {
                    status = "disabled";
            };
           lsm6ds3: lsm6ds3@0 {
				spi-max-frequency = <500000>;
				compatible = "st,lsm6ds3";
                                reg = <0x0>;
			};

		};
	};
};

```

closed 2020-04-14 by Panky-codes