STMicroelectronics/STMems_Linux_IIO_drivers #3: Why the supported ODR for st_lsm6ds3 IMU is only until 416 Hz?

Panky-codes, 2020-10-01:
This is more of a question than an issue. I initially wrote a userspace driver for lsm6ds3 but later found at that there is already an kernel IIO driver. 
But after some investigation, I saw that the kernel driver `st_lsm6ds3` only supports ouput data rate (ODR) upto only 416 Hz as it can be seen in this struct definition in `drivers/iio/imu/st_lsm6ds3/st_lsm6ds3_core.c`:
```c
static struct st_lsm6ds3_odr_table {
	u8 addr[2];
	u8 mask[2];
	struct st_lsm6ds3_odr_reg odr_avl[ST_LSM6DS3_ODR_LIST_NUM];
} st_lsm6ds3_odr_table = {
	.addr[ST_MASK_ID_ACCEL] = ST_LSM6DS3_ACCEL_ODR_ADDR,
	.mask[ST_MASK_ID_ACCEL] = ST_LSM6DS3_ACCEL_ODR_MASK,
	.addr[ST_MASK_ID_GYRO] = ST_LSM6DS3_GYRO_ODR_ADDR,
	.mask[ST_MASK_ID_GYRO] = ST_LSM6DS3_GYRO_ODR_MASK,
	.odr_avl[0] = { .hz = 13, .value = ST_LSM6DS3_ODR_13HZ_VAL },
	.odr_avl[1] = { .hz = 26, .value = ST_LSM6DS3_ODR_26HZ_VAL },
	.odr_avl[2] = { .hz = 52, .value = ST_LSM6DS3_ODR_52HZ_VAL },
	.odr_avl[3] = { .hz = 104, .value = ST_LSM6DS3_ODR_104HZ_VAL },
	.odr_avl[4] = { .hz = 208, .value = ST_LSM6DS3_ODR_208HZ_VAL },
	.odr_avl[5] = { .hz = 416, .value = ST_LSM6DS3_ODR_416HZ_VAL },
};
```
But the actual hardware supports ODR for accelerometer upto 6.66 kHz and for gyro upto 1.66 kHz. Is there a reason why the support for this ST chip is only until 416 Hz? Is there a way to scale it up? 

Any help is appreciated!

mariotesi, 2020-10-02:
Hi,

This drivers repository are targeted for personal electronics devices and mass market purpose, so I think 6.6 kHz/1.6 kHz are not in the scope also because on vast majority of cases these drivers are used with Android OS where the requested max ODR in CDD specs is 400 Hz.
Moreover too much interrupt per seconds (when no batching) on some target boards would lead to excessive CPU workload, this would create problems in the management of customer support requests.
Anyway as you know these drivers are released here with the "as is" formula, so they can be changed easily as needed by extending the st_lsm6ds3_odr_table array and updating the ST_LSM6DS3_ODR_LIST_NUM values.

Best Regards,
Mario

Panky-codes, 2020-10-03:
Thanks Mario. That explains the reason! Will do as you suggested!

closed 2020-10-03 by Panky-codes