STMicroelectronics/STMems_Linux_IIO_drivers #5: How to get access to the buffer and triggers with the st_imu68 IIO driver?

Ghlens, 2020-12-18:
I am trying to use an hr timer with the st_imu68_i2c driver and the LSMDS1 IMU sensor. But even though I have the CONFIG_IIO_BUFFER enabled, the scan_elements, trigger, and buffer directory are missing.

Our config contains the following IIO configuration:

```
# CONFIG_TOUCHSCREEN_TSC2007_IIO is not set
# CONFIG_SENSORS_IIO_HWMON is not set
# IIO staging drivers
CONFIG_IIO=y
CONFIG_IIO_BUFFER=y
CONFIG_IIO_BUFFER_CB=y
CONFIG_IIO_KFIFO_BUF=y
CONFIG_IIO_TRIGGERED_BUFFER=y
CONFIG_IIO_CONFIGFS=y
CONFIG_IIO_TRIGGER=y
CONFIG_IIO_CONSUMERS_PER_TRIGGER=2
CONFIG_IIO_SW_DEVICE=y
CONFIG_IIO_SW_TRIGGER=y
# CONFIG_IIO_ST_ACC33 is not set
# CONFIG_IIO_ST_LIS3DHH is not set
# CONFIG_IIO_ST_ACCEL_3AXIS is not set
# CONFIG_IIO_ST_LIS2DS12 is not set
# CONFIG_IIO_ST_ISM303DAC_ACCEL is not set
# CONFIG_IIO_ST_LIS2HH12 is not set
# CONFIG_IIO_ST_LIS2DW12 is not set
# Hid Sensor IIO Common
# CONFIG_IIO_SSP_SENSORHUB is not set
# IIO dummy driver
# CONFIG_IIO_SIMPLE_DUMMY is not set
# CONFIG_IIO_ST_GYRO_3AXIS is not set
# CONFIG_IIO_ST_LSM6DSX is not set
# CONFIG_ST_LSM6DSM_IIO is not set
# CONFIG_IIO_ST_LSM6DSO is not set
# CONFIG_ST_LSM6DS3_IIO is not set
# CONFIG_ST_LSM6DS3H_IIO is not set
# CONFIG_ST_ISM330DLC_IIO is not set
CONFIG_IIO_ST_IMU68=y
CONFIG_IIO_ST_IMU68_I2C=y
CONFIG_IIO_ST_IMU68_SPI=y
# CONFIG_IIO_ST_ASM330LHH is not set
# CONFIG_IIO_ST_LSM6DSR is not set
# CONFIG_IIO_ST_ISM330DHCX is not set
# CONFIG_IIO_ST_LSM6DSO32 is not set
# CONFIG_IIO_ST_MAGN_3AXIS is not set
# CONFIG_ST_MAG40_IIO is not set
CONFIG_ST_MAG3D_IIO=y
CONFIG_ST_MAG3D_I2C_IIO=y
CONFIG_ST_MAG3D_SPI_IIO=y
# CONFIG_IIO_MUX is not set
CONFIG_IIO_HRTIMER_TRIGGER=y
CONFIG_IIO_INTERRUPT_TRIGGER=y
CONFIG_IIO_TIGHTLOOP_TRIGGER=y
CONFIG_IIO_SYSFS_TRIGGER=y
# CONFIG_IIO_ST_PRESS is not set
# CONFIG_ST_LPS22HB_IIO is not set
# CONFIG_ST_LPS33HW_IIO is not set
# CONFIG_ST_LPS22HH_IIO is not set

```
 I only got the directories to appear after I added an interrupt to the device tree like seen below. This however is a dummy interrupt as we do not have an actual physical wire connected to the IMU to the SOM.

```&i2c1 {
	lsm9ds1@6b {
		compatible = "st,lsm9ds1";
		reg = <0x6b>;
                interrupt-parent = <&gpio1>;
                interrupts = <5, IRQ_TYPE_LEVEL_HIGH>;
		status = "okay";
	};

	lsm9ds1_magn@1e {
		compatible = "st,lsm9ds1_magn";
		reg = <0x1e>;
                interrupt-parent = <&gpio1>;
                interrupts = <5, IRQ_TYPE_LEVEL_HIGH>;
		status = "okay";
	};
};
```
Is there something I am doing wrong? Or does the st_im68 driver only support hardware interrupts?

mariotesi, 2021-01-04:
Hi, 

As you can see here:
https://github.com/STMicroelectronics/STMems_Linux_IIO_drivers/blob/linux-4.4.y-gh/drivers/iio/imu/st_imu68/st_imu68_core.c#L583

The driver allocate the buffer if an interrupt is defined into the device tree, otherwise you nned to go in polling to the raw read data

Ghlens, 2021-01-04:
Thank you for your help. This confirms the way I thought the driver works.

closed 2021-01-04 by Ghlens