STMicroelectronics/STMems_Linux_IIO_drivers #8: Running the driver on Android 10

erhan14, 2021-05-05:
We are using AOSP (Android 10). Kernel version is 5.4.47
imu/lsm6dsm driver in 4.19 branch ported to Linux 5.4.47
STMems_Android_Sensor_HAL_IIO/ HAL integrated into Android 10

05-05 13:02:43.222   317   317 D SensorHAL: 10 IIO devices available into /sys/bus/iio/devices/ folder.
05-05 13:02:43.223   317   317 D SensorHAL: "lsm6dsl_step_d": IIO device found and supported. Wake-up sensor: no
05-05 13:02:43.223   317   317 D SensorHAL: "lsm6dsl_sign_motion": IIO device found and supported.
05-05 13:02:43.224   317   317 D SensorHAL: "lsm6dsl_accel": IIO device found and supported. Wake-up sensor: no
05-05 13:02:43.255   317   317 D SensorHAL: "lsm6dsl_tilt": IIO device found and supported.
05-05 13:02:43.257   317   317 D SensorHAL: "lsm6dsl_step_c": IIO device found and supported. Wake-up sensor: no
05-05 13:02:43.258   317   317 D SensorHAL: "lsm6dsl_gyro": IIO device found and supported. Wake-up sensor: no
05-05 13:02:43.296   317   317 D SensorHAL: "LSM6DSL Step Detector Sensor": injection mode not available.
05-05 13:02:43.296   317   317 D SensorHAL: "LSM6DSL Step Detector Sensor": created HW class instance, handle: 1 (sensor type: 18).
05-05 13:02:43.296   317   317 D SensorHAL: "LSM6DSL Significant Motion Sensor": injection mode not available.
05-05 13:02:43.296   317   317 D SensorHAL: "LSM6DSL Significant Motion Sensor": created HW class instance, handle: 2 (sensor type: 17).
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Accelerometer Sensor": injection mode not available.
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Accelerometer Sensor": created HW class instance, handle: 3 (sensor type: 1).
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Tilt Sensor": injection mode not available.
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Tilt Sensor": created HW class instance, handle: 4 (sensor type: 22).
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Step Counter Sensor": injection mode not available.
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Step Counter Sensor": created HW class instance, handle: 5 (sensor type: 19).
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Gyroscope Sensor": injection mode not available.
05-05 13:02:43.297   317   317 D SensorHAL: "LSM6DSL Gyroscope Sensor": created HW class instance, handle: 6 (sensor type: 4).
05-05 13:02:43.297   317   317 D SensorHAL: "Accelerometer Uncalibrated Sensor": created SW class instance, handle: 7 (sensor type: 35).
05-05 13:02:43.299   317   317 D SensorHAL: Dynamic Sensor Manager: created Dynamic Sensor Proxy, handle: 8 (sensor type: 32).
05-05 13:02:43.299   317   317 D SensorHAL: 8 sensors available and ready.

When we want to test it with sample APKs in the market the following log is output:

[  307.355780] st-lsm6dsm-i2c 2-006b: data fifo overrun, failed to read it.

Can you send us what can cause this and steps needed to resolve this?

mariotesi, 2021-08-18:
Hi,

I think the interrupt pin is not configured/connected to the AP. Please try to read interrupt proc fs:

cat /proc/interrupts

and check counter of lsm6dsl interrupt

closed 2022-10-14 by mariotesi

SyedAshja, 2023-01-10:
@mariotesi hey did you ever find the solution?