STMicroelectronics/STMems_Linux_IIO_drivers #9: How to save and read a lot of data at once?

simonko-plainy, 2021-05-24:
I use ASM330LHH as Linux IIO.
Source and DTS were added.
We found that /sys/bus/iio also added a device.
I also checked reading data in direct mode from sysfs.
Therefore, I think there is no problem with the H/W connection.

I try to read the value of asm330lhh every 15mS for collision detection.
I want to fetch the collected data rather than reading the value every time.
I want to know how to save a lot of data automatically and read it at once.
How should I use it?
​

mariotesi, 2021-08-18:
Hi,

You can use the generic_buffer tool and configure the sample rate of ASM330LHH to 104 Hz in order to guarantee the requested 15 ms.
You can also set the FIFO buffer to max value (416) by echo 416 > /sys/bus/iio/devices/iio:device<x>/hwfifo_watermark, in such way do not to have an interrupt at each sample but only when the FIFO is full and then read the FIFO data in a single shot with generic_buffer

closed 2021-10-04 by mariotesi

AlysonQ, 2024-05-03:
Hi @mariotesi ,
I used to generic_buffer too to read asm330hhxg1_gryo's data.
but whatever I tried, nothing shows.
Could you help me to solve this problem?

```bash
# cat  /sys/bus/iio/devices/iio\:device1/sampling_frequency
833.000000
# cat  /sys/bus/iio/devices/iio\:device1/name 
asm330lhhxg1_gyro
#  ./iio_generic_buffer -a -n asm330lhhxg1_gyro -e -g
iio device number being used is 1
trigger-less mode selected
No channels are enabled, enabling all channels
Enabling: in_anglvel_z_en
Enabling: in_timestamp_en
Enabling: in_raw_en
Enabling: in_anglvel_y_en
Enabling: in_anglvel_x_en
nothing available
nothing available
Disabling: in_anglvel_z_en
Disabling: in_timestamp_en
Disabling: in_raw_en
Disabling: in_anglvel_y_en
Disabling: in_anglvel_x_en
# 
```