STMicroelectronics/STMems_Linux_IIO_drivers #18: ASM330LHH does not generate interrupt on my board

arleywong, 2023-04-18:
At my side, I configured DTS as following. 
&hsi2c_10 {
       status = "okay";
               
       #address-cells = <0x1>;
       #size-cells = <0x0>;
       asm330lhh@6a {
               compatible = "st,asm330lhh";
               reg = <0x6a>;
               interrupt-parent = <&gpp5>;
               interrupts = <5 IRQ_TYPE_EDGE_RISING>;
       };
};

As a result:

1) I2C operations are all good, and I am able to directly Get/Set registers by using I2C commands.
Command: /usr/sbin/i2cget -f -y -a 10 0x6a 0x0F
Return: 0x6b
Command: cat /sys/bus/iio/devices/iio\:device0/in_anglvel_x_raw
Return: 57

2) However, when I use libiio (command: iio_readdev asm330lhh_gyro anglvel_x), it always goes to timeout
There is no GPIO interrupt generated at all. I belive this leads to timeout.
This is my question which I want to get help. Could you help or give me some hint? Thank you very much!

3) One wierd thing is, when I execute following two commands, one GPIO interrupt will be generated and IRQ handler of ASM330LHH driver is able to read one pair of FIFO data.
/usr/sbin/i2cset -f -y -a 10 0x6a 0x12 0x44;
/usr/sbin/i2cset -f -y -a 10 0x6a 0x12 0x64;

mariotesi, 2023-04-18:
Please double check on /proc/interrupts for existing id asm330lhh,
something like:
# cat /proc/interrupts  | grep asm330

if it exists then try with timeout set to 0:
# iio_readdev -T 0 -b 1 asm330lhh_accel -s 100

arleywong, 2023-04-19:
> 

Dear Mariotesi,

Thanks for your reply!!
Could you guide where to find the repo driver?

1. At my side, following command returns nothing.
```shell
cat /proc/interrupts | grep asm330
``` 

2. Searched "asm330lhh" string in kernel code directory, however, st_lsm6dsx is the only result.
```c
user@server:kernel$ grep asm330lhh  drivers/ -nr
drivers/iio/imu/st_lsm6dsx/Kconfig:14:    ism330dlc, lsm6dso, lsm6dsox, asm330lhh, lsm6dsr, lsm6ds3tr-c,
drivers/iio/imu/st_lsm6dsx/st_lsm6dsx_i2c.c:65:         .compatible = "st,asm330lhh",
drivers/iio/imu/st_lsm6dsx/st_lsm6dsx_spi.c:65:         .compatible = "st,asm330lhh",
drivers/iio/imu/st_lsm6dsx/st_lsm6dsx.h:24:#define ST_ASM330LHH_DEV_NAME        "asm330lhh"
```

mariotesi, 2023-04-19:
I presume it depends on irq configuration, please check interrupt pin is connected to your target board application processor, furthermore consider the interrupt configuration of device tree is related to your hardware platform, for instance on Raspberry pi 4 I use the following two overlays for SPI-0 and I2C respectively:

// i2c
        fragment@0 {
                target = <&spi0>;
                __dormant__ {
                        asm330lhhx: asm330lhhx@0 {
                                compatible = "st,asm330lhhx";
                                spi-cpol;
                                spi-cpha;
                                reg = <0>;
                                spi-max-frequency = <10000000>;
                                interrupt-parent = <&gpio>;
                                interrupts = <13 IRQ_TYPE_LEVEL_HIGH>;
                                st,int-pin = <1>;
                                status = "okay";
                        };
                };
        };

        __overrides__ {
                addr = <&asm330lhhx>,"reg:0";
                intpin = <&asm330lhhx>,"st,int-pin:0";
                interrupt = <&asm330lhhx>,"interrupts:0";
                asm330lhhx = <0>, "+0";
}

// spi
        fragment@0 {
                target = <&i2c_arm>;
                __dormant__ {
                        #address-cells = <1>;
                        #size-cells = <0>;
                        status = "okay";

                        asm330lhhx: asm330lhhx@6b {
                                compatible = "st,asm330lhhx";
                                reg = <0x6b>;

                                interrupt-parent = <&gpio>;
                                interrupts = <26 IRQ_TYPE_LEVEL_HIGH>;

                                st,int-pin = <1>;
                                drive-pullup-shub;
                                status = "okay";
                        };
                };
        };

        __overrides__ {
                addr = <&asm330lhhx>,"reg:0";
                intpin = <&asm330lhhx>,"st,int-pin:0";
                interrupt = <&asm330lhhx>,"interrupts:0";
                asm330lhhx = <0>, "+0";
}

Another check that you could do once the sensor is activated with iio_readdev / generic_buffer is to verify that the interrupt pin of the sensor is set (int pin number of sensor depends on "st,int-pin" property value set in device tree, default is int 1), in this case if the handler doesn't trigger it is a matter of the miss configuration of GPIO interrupt pin on the target board BSP.

arleywong, 2023-04-19:
@mariotesi 

I think you are right, this very much relates to configuration with board BSP. It is just I have no knowledge about it and there is no specific clue.
Anyway, I will check further. Thank you so much!!!

Besides, there is one info. for your reference.
**If "drive-open-drain;" is added to dts, the data-ready interrupt will be reported endlessly just as system booting (I do not executed any iio command), and interrupt handler is called repeatedly (later it exausts all cpu and leads to crash).**
So I guess:

- the physical interrupt line is connected OK.
- high probably it is dts configuration issue.

This is the DTS node and the endless interrupt statistics:

```c
&hsi2c_10 {
       status = "okay";

       asm330lhhx-imu@0x6a {
               compatible = "st,asm330lhh";
               reg = <0x6a>;
               interrupt-parent = <&gpp5>;
               interrupts = <5 IRQ_TYPE_LEVEL_HIGH>;
	       drive-open-drain;
               st,int-pin = <1>;
               st,mlc-int-pin = <2>;
               mount-matrix = "1", "0", "0",
                       "0", "1", "0",
                       "0", "0", "1";
       };
};
```

arleywong, 2023-04-19:
Moreover, if "drive-open-drain" is added as following, after device probe is done, the data-ready interrupt will also be reported endlessly and interrupt handler is called repeatedly.
This time, no need to execute i2cset commands at all.

&hsi2c_10 {
       status = "okay";

       asm330lhhx-imu@0x6a {
               compatible = "st,asm330lhh";
               reg = <0x6a>;
               interrupt-parent = <&gpp5>;
               interrupts = <5 IRQ_TYPE_LEVEL_HIGH>;
			   drive-open-drain;
               st,int-pin = <1>;
               st,mlc-int-pin = <2>;
               mount-matrix = "1", "0", "0",
                       "0", "1", "0",
                       "0", "0", "1";
       };
};

arleywong, 2023-04-23:
@mariotesi 
Finnally, H/W part said there is H/W issue on interrupt pins.
So I wil close this issue.
Thank you!

closed 2023-04-23 by arleywong